cyberboy666/r_e_c_u_r #10: some keys on the num pad are not usable on device

langolierz, 2018-01-03:
the numlock key and the 000 key have a specific operation that makes it difficult to map within the program. a custom keyboard layout will need to be created on machine to strip these functions and make it easier.

leobrowning92, 2018-01-12:
key remappings were achieved using the program `xmodmap`, and the most helpful post
you can find the current keymappings using `xmodmap -pke` and for the specifically numpad related keymappings use:

    xmodmap -pke | grep KP > defaultKeymap
    xmodmap -pke | grep Nu >> defaultKeymap

which prints only the default numpad keys to a file for reference later.
keys can be manually remapped using `xmodmap -e "keycode 82=a" for testing.

In order to have the custom keymap work on startup we have added the line `xmodmap ~/.remap`

where the file `.remap` has a list of the keycodes that you want to include.

TODO add the `.remap` file to the project directory once complete

leobrowning92, 2018-01-14:
the info aboce was added to the dotfiles readme. the .remap file still needs to be added to the dotfiles folder. 

after that is done, I think that this issue can be closed, as the ongoing problem with the 000 key is being handled in issue #29

langolierz, 2018-01-14:
sounds good - i will do this and close issue

closed 2018-01-15 by langolierz

sym3tri, 2021-03-13:
Hi. I'm a newb here and looking for help debugging my janky keypad that's not working. I understand the above description, but I can't find any docs on where to actually place the remap file. I guess I would have to rebuild everything from source just to get the keys to remap... is that how it works?