cyberboy666/r_e_c_u_r #35: load key mappings from json

langolierz, 2018-01-15:


leobrowning92, 2018-01-15:
This is the mapping from the a,b,c to program functions which will be easily user configurable right?

langolierz, 2018-01-16:
yupp thats the one - any ideas of how to structure it ? thinking would want a (?) list of keys, each one containing (dict ?) of actions for main press and alt-press. also want to capture the option to have keys with different uses from different display modes (ie up moves navigation bar up by default but seeks the video back in SAMPLER mode...)

langolierz, 2018-01-17:
the format i am going for looks like this: 

`{
	"a": {
                "DEFAULT": [this_is_used_if_no_MODE_overrides,this_this_used_for_alt_function]
		"BROWSER": ["move_browser_selection_up"],
		"SAMPLER": ["seek_back_on_player"],
		"SETTINGS": ["move_settings_selection_up"]
	},
	"b": {
		"BROWSER": ["move_browser_selection_down"],
		"SAMPLER": ["seek_forward_on_player"],
		"SETTINGS": ["move_settings_selection_down"]
	},`

langolierz, 2018-01-17:
everything is working here as intended except 000 key is not useable as described in #29

closed 2018-01-17 by langolierz