cyberboy666/r_e_c_u_r #36: duplicate usb folders loaded

langolierz, 2018-01-17:
sometimes the pi loads my usb as 'TIM1' and has a 'TIM' folder stuck in pi/media.

langolierz, 2018-04-03:
this seems to happen when a usb is removed or inserted at after or around boot. with some poking i can probably reproduce it. it can be fixed by deleting the folders mounted. i could make this an action from inside the program if it continues to be a problem...

langolierz, 2018-09-24:
to fix this problem just type `sudo rm -r /media/pi/**name_of_device**`

TubularCorpration, 2020-09-08:
Just a heads up I was experiencing this problem and your solution worked but it also wiped the filesystem of the actual USB stick! I confirmed on a Windows machine, it's empty now.  I have everything backed up so not a big deal but if I didn't it would have been a hassle recovering everything.

langolierz, 2020-09-08:
oh god @TubularCorpration im sorry this happened ! (and lucky you had it backed up!)

i meant to type this with the device _removed_ first. will edit my previous comment to make it clear - thanks for the heads up (you prob saved someone a lot of trouble ! 😅 )

TubularCorpration, 2020-09-08:
Wasn't an issue at all, I just copied everything back on to the USB stick, but I wanted to post about it in here to make sure nobody without backups made the same mistake!