cyberboy666/r_e_c_u_r #63: Add tags to make this easily discoverable on github

leobrowning92, 2018-08-09:
Just realised how important tags are on github when people are searching for repositories.
Would be a good idea to put them in if you are interested in this :)

langolierz, 2018-09-24:
@leobrowning92 from what i can see , tags are directly related to releases , which i havent been using , unless im missing something ? i will create a release for upcoming features soon , so could try adding tags to that ?

leobrowning92, 2018-10-05:
hi, sorry it took a while get back to you. Ooh there is a duplication of terminology here. There are tags for a github repository, which make searching it on github easier. There are also tags which are a feature of git, used to mark certain commits. This issue relates to the first case, tags on github to make things more searchable.

as an aside, github has releases which you mention that can be linked to a certain tag on a commit, or directly to a commit. I have found it very useful to track large code releases in my bigger projects, even just for myself. I have found its a great way to link to output from simulations or generative projects that is a bit coarser then using commits.

langolierz, 2018-10-18:
hah, i cant figure out how to tag the repo (different from a release) and all the info online is referring to the later, can you point me to where i would do this @leobrowning92 ?

langolierz, 2018-11-10:
dun !

closed 2018-11-10 by langolierz