cyberboy666/r_e_c_u_r #73: create a pcb for simple incur circuit

langolierz, 2018-10-19:
making it easier for people to use continuous inputs from the gpio by creating a pcb that fits the current enclosure and list the parts and details on how to wire it up (with pics etc)

langolierz, 2018-12-18:
this is in progress with first revision on way from fab house. still need to publish a BOM, also got idea for a v2 board that attaches to the pi kinda like a hat would ...

sebiik, 2019-05-06:
My PCB fab house informed me that the gerbers are incomplete, specifically there is "no bottom solder mask layer" present.
I happen to have experience with DIY synth/eurorack/video circuits and could help out with the PCB design. Any interest?

langolierz, 2019-05-07:
@sebiik yess ; oh i forgot i even had gerbers on this repo, since then i have been learning kicad and made some better proto-pcb's for this - just havnt updated here yet. 

my first tries had some mistakes in them. the current version is working for analog-ins atleast. i havnt had the chance to test with din-midi yet - as soon as that is confirmed i will upload these improved gerbers.

i am alway interested in help ! email me and we can talk about circuits

langolierz, 2019-08-06:
ths has been updated - schemaic and gerber and a build guide /bom is started

closed 2019-08-06 by langolierz