cyberboy666/r_e_c_u_r #77: template and guide to writing first shaders

langolierz, 2018-10-19:
this is partially already documented in conjur repo , but might be a bit highlevel for someone interested in writing their first shader, it should include:

- how to set up a shader deving environment (using glslviewer etc)
- template shader explaining the basics , lines to uncomment and see the changes etc
- basic concepts for creating simple gen-shaders
- template shader for texture input (colour and position) with basic concepts for creating simple pro-shaders
- maybe how to dev on ur mac/windows machine ?

sebiik, 2019-05-06:
I'm curious where you'd want to set up the shader deving environment. Add an option in recur to start glslviewer mode on the pi so you could connect a USB keyboard and develop from there?

langolierz, 2019-05-07:
@sebiik  , yeah - i was thinking basically some instructions on how to edit a shader on the pi, (probably dont need a option to start glslviewer from within recur at first although that would be cool!),
something like:

- plug in a usb-keybboard
- press `.` to exit recur
- open termit from start-menu, type `glslViewer ~/Shaders/my_new_shader.frag -l`
- open my_new_shader.frag in gedit and try changing these values ...
- do this to load in a image-texture to be processed , try changing these values ... 

do you have any ideas on how to ease the entry for people customising there own video-effects etc ?

averagehat, 2020-01-10:
If using ssh #110 works well; 

One thing about just using glslViewer is that you have to supply something for the (midi) param values or else a lot of the shaders won't really work.

By being connected to the pi, I can use my own editing environment on a different machine (and even use glslViewer there as a single-shader preview, though there is still the param values problem) and get good live feedback for learning, especially helpful with the 2-input shaders which I don't even know how to create with glslViewer or anything else.

I will make a special effort to document my learning and if it works out I can put together some documentation about editing environment, ssh setup, glsl, whatever.

There is already a bit of documentation and templating here: 
https://github.com/langolierz/r_e_c_u_r/blob/9935f1a98bf308b1563e54ca8551ac16854bda9e/Shaders/2-input/mix_lumaKey.frag

konsumer, 2024-09-12:
On this angle, I added these (discussed in #152 )

- [glsl_sandbox](https://github.com/konsumer/glsl_sandbox) - basically just some knobs for native glslviewer.
- [recurweb](https://github.com/konsumer/recurweb) a web-based editor, that has support for video and inputs. (visit [here](https://konsumer.js.org/recurweb/))

While working on these I also:
- expanded, normalized & cleaned-up [shader collection](https://github.com/konsumer/glsl_sandbox/tree/main/shaders)
- created a [simple template](https://github.com/konsumer/glsl_sandbox/blob/main/shaders/template.frag)

Feel free to use these here (let me know, and I will make a PR for shaders.)