cyberboy666/r_e_c_u_r #79: using a 3.2' display

hacklab01, 2019-03-10:
Hi, I have a 3.2' pi display. 

Editing some stuff in Pi config files of the pre installed image, the display seems to work but the interface is all messed up. 

Any way to make it fit into a 3.2' 320 x 240 px display?

Tnx! :)

hacklab01, 2019-03-10:
OK I kind of fixed by changing the font size but some documentation on how ~/r_e_c_u_r/display_centre/display.py is setup might help. :)

timcaldwell-kitchenstories, 2019-03-11:
hey @hacklab01  - cool, i was going to suggest changing the font-size. what size did you set and how does it look ? also what was the config file you changed ? i can add to the faq about 3.2' displays if others want to try it.

and yeah , i know the display code is not that readable and quite 'hard-coded' to that particular screen. i should write a little bit in the develop doc page about it. if you have any suggestions on what would be useful to include lemme know too

ps hacklab01 sounds cool :)

hacklab01, 2019-03-11:
Hi, thanks for the speedy reply. 

I changed the font size [here](https://github.com/langolierz/r_e_c_u_r/blob/fa92acca6359fa8a2431509ecdece405816e06a5/display_centre/display.py#L30)  to 7 but now the ratio is now off. I'll send pics as soon as I reconnect the setup on my desk.

Hehe yeah, tnx, hacklab01 is hackerspace I'm a part of. We have an open call for hacking (or not) related stuff so if anyone is interested in presenting a project / holding a workshop or similar and getting some travel funds / accommodation / pocket money feel free to apply https://hacklab01.org/hackn3t (shameless self promotion).

timcaldwell-kitchenstories, 2019-03-11:
yes pls. wow sounds cool ! i have this and some other upcoming video-hardware related projects  that could be workshops / talks soon. would be nice to visit !

langolierz, 2019-03-11:
whoops commenting on my work account lol. yeah pls send pics of how it looks, i will update the docs and close then close this issue if it looks okay.

and wow ! sounds cool , i have this and some upcoming diy-video projects that could be workshops / talks soon maybe. would be nice to visit !

hacklab01, 2019-03-11:
I mean it's usable I guess (I tried fiddling around with for a bit, didn't get accustomed to the controls yet), just the ratio is really off and there's a lot of screen space empty which could be filled... OK, yeah I'll procrastinate other stuff by sending pics of a raspberry Pi... Why not... =D

Here we go https://imgur.com/a/5Z4O4hE
As you can see I added the same video several times because n00b.

Also it looks like either I'm not getting something or the controls on my numpad are a bit off (it looks like the first row is shifted to the left by one key). Not sure how that happened, I guess I could just remap it by hand, also not sure why that would help (all numpads are the same? Or not?). I'll follow up on that one shortly.

hacklab01, 2019-03-11:
The numpad thing was pretty easy - not all numpads are created equal - mine has the backpspace keys and the minus key switched out. :)

hacklab01, 2019-03-11:
Here's a pic of the numpad https://imgur.com/a/R7MmZbu I don't know the exact model the sticker on the back is scraped off. :) 

Yeah i kinda missed the expression, didn't mean the ratio, ment that it's not expanded to screen size. Yeah - If I use a float value for the font size the GUI doesn't load... 

I just realized that I missed the pad configuration again haha is it cab or bac or acb noone knows. =D

Another question: Is there a better way to restart the system without restarting the whole pi? Killing python and then running ~/r_e_c_u_r/r_e_c_u_r.py sometimes works and sometimes it just produces a black screen until reboot.

closed 2019-12-08 by langolierz