cyberboy666/r_e_c_u_r #80: looping buggy?

hacklab01, 2019-03-11:
It's me again - I'm pretty sure this is just me but looping isn't really working for me. 

I set in and out on pretty any video I try and get a really jittery erratic experience, a black screen and even lock ups until I clear the deck. 

Is there a specific codec omxplayer likes better for looping? My vids are h264 1920x1080 

Could it be the SD card. The card is a new Sandisk card and copy speeds seem fine while testing moving some files around. 

Any tips?

hacklab01, 2019-03-11:
I mean, yeah - obviously - the more keyframes you have in your video the prettier the loops are.
If the video doesn't have much keyframes, omxplayer is gonna break while trying to make a loop. :( 
Maybe mention something about encoding in the manual? IDK... :)

hacklab01, 2019-03-11:
Just to leave this here if anyone has the same issues as me - it's ideal to have a keyframe on every frame (that creates a larger file but the days of caring about filesize are behind us). The looping then works perfectly, well _almost_. :) 

You can just make ffmpeg output every frame as a keyframe with

`ffmpeg -i input.file -q:v 0 -q 1 output.file`

It will select x264 as the codec by itself. -q:v 0 is OK quality for me, if you need better use -crf 17 instead but that makes the file even bigger.

langolierz, 2019-03-12:
hey @hacklab01 ; thanks for the detailed information and suggestion about adding keyframes. 

in my experience testing this , i did find some issues similar to what you describe especially for 1080 videos. there is a setting in the menu called 'HDMI_MODE' , which when set to 'CEA 4 HDMI' should drop the output resolution to 720 but handle playing 1080 videos a bit better. and yes i think adding more key frames also helps for seeking to the beginning of the subloop. i will update the faq with this infomation you have included here.

if you felt like sending me one of the files you were using (before adding keyframes) i could test it on my recur to see if we experience the same thing

hacklab01, 2019-03-13:
Hey @langolierz  I'm using RCA output so the HDMI swiitch thing won't help. The video isn't 1080p, I put both files here https://v3d.space/files/ you can download and check. The video is several scenes from  https://archive.org/details/InvasionOfTheBeeGirls . Just pls ping me when you download them so I can kill the directory, because I doubt the movie is really public domain. =D

langolierz, 2019-03-14:
i have downloaded the files so you can bring them down. hmmm.

to reset the program you should be able to stop recur (pressing '.' or maybe fn + 9 ?) ; then start again by running `python3 ~/r_e_c_u_r/r_e_c_u_r.py ` ; having to reclone the repo sounds like a problem with the persisted data - rather than reclone i think you can just delete `json_objects/display_data.json` and/or
`json_objects/next_bankslot_number.json` ;

i assume you are using the v1-image , you might want to try v2beta which can be found here https://github.com/langolierz/r_e_c_u_r/blob/c_o_n_j_u_r/documentation/beta_access_notes_and_limitations_of_v2.md , some bugs might be fixed , then again there might be more. i know that there is a `reset` option in the settings under other atleast in v2beta .

i still think maybe the 1080p files are part of the problem you are having. maybe you could try with a few sd videos and see if the same thing happens ? i will see if i can reproduce these problems soon too

 (ps soz you are experiencing so many bugs ; i hope its still interesting or useful ... )

hacklab01, 2019-03-14:
Hmm, I tried deleting the contents of the json directory cause I figured stuff was there but it didn't help once so I just resorted to cloning. 

I'll use the v2 image and report. 

The file I sent you isn't 1080p it's an SD file and it still has issues. It doesn't really matter tho, reencoding with more keyframes is simple enough. 

I don't mind the bugs at all - I actually had an idea to build a numpad controlled video sampler / player (mine wouldn't have a screen tho haha), so I'm more then happy to help with this as it does exactly what I needed and more.

hacklab01, 2019-03-15:
The beta works much nicer with the openframeworks backend. SD vids loop just fine regardless of keyframes. 

I think we can close this and attribute the lockups to omxplayer and HD videos. If you have nothing to add, please close this issue. 

The solution is to use sd videos and  the openframeworks backend.

closed 2019-07-16 by langolierz