cyberboy666/r_e_c_u_r #85: Improve some of the 2-input shaders

langolierz, 2019-08-21:


averagehat, 2020-01-14:
@langolierz  Any particular ideas? I would be interested in working on this.

One thing would be using more parameters (for example, wipe.frag only uses 2 out of four parameters).

langolierz, 2020-01-14:
awesome ! a few things come to mind quickly, one would be to make the
lumakey do black keying and white keying (maybe using one of the unused
params as a 'switch', as i tried to do for a->b, b->a switch ...) , or they
could be separate shaders,

another is to create a chromakey shader, one param for key amount, maybe
another (2? to choose the colour, or ... something cant remember how now
but when i had an mx50 i remember the chromakey being fun ...)

yes you could add more to the wipe one , or add more wipe shaders ,
diagonal , circles, multi, etc

i think i wrote this issue originally because i suspected the lumakey
shader wasnt behaving exactly how i expected , maybe my implementation has
a mistake, or maybe theres ways to make it smoother in general ...

just some ideas , its a pretty open question , but i agree that the 2input
shaders are really fun to play with !