cyberboy666/r_e_c_u_r #91: Feature Request: NDI Support

teafella, 2019-10-18:
It would be great to be able to feed recur [NDI](https://ndi.tv/about-ndi/)

This would eliminate the need for a capture card when trying to get video out of your computer (or other pis) into recur.

It would also enable you to feed recur further into a VJ setup using Resolume.

Would this be difficult to do?

langolierz, 2019-10-18:
i thought about this too. there is not much info online about ndi on rpi's , i think it is only recently even compatible with ARMs.  if possible it would be through the openframeworks extension ofxNDI , this thread sounds promising - although only sending not receiving , maybe possible in the near future ? https://github.com/leadedge/ofxNDI/issues/4

langolierz, 2019-10-18:
https://forums.newtek.com/showthread.php/151197-NDI-for-arm-devices

teafella, 2019-10-18:
I have a C++ class for sending NDI if you are interested, I'm not sure how easy it is to include in the python codebase though. NDI has a specific shared binary for rpi at /lib/ndi/arm-rpi3-linux-gnueabihf/ on the linux lib 

Idk how this works with ofx since ive never used it. From what I understand its C++ based so it should be possible.

I haven't spent time properly testing the receiving code yet but it seems to be working based on the test code in the library.

langolierz, 2019-10-21:
cool ! that sounds promising , the r_e_c_u_r backend is based in openframeworks, which yeah is just libraries around C++ , so it should be possible. i will have to look into this and have a play around. im not much of a C++ dev though so lets see. if you get receiving code working in C++ i would be keen to have alook

magdesign, 2019-10-24:
Sienna said they can do it:
https://twitter.com/NDI_Cloud/status/1145754510343778314?s=19
but nothing to find. Asked the support and they said "no known release date...."

teafella, 2019-11-21:
I tried implementing a receive on the PI and I am only able to receive audio at the moment, which leads me to believe the SDK doesn't support video, I am contacting ndi via email to see whats up since a new version of NDI was just released.

From what I gather from the forums, the issue is lack of ndi codec for arm based processors.

Re Sienna:
Their iPad app (ipad is arm based) also seems to receive NDI, though its very laggy and freezes after like 5 min. I'm not sure how they where able to hack this together...

teafella, 2019-11-21:
Reply from NDI support:
"At the current time we do not have an optimized version of the ARM decoder. This is something we would like vey much to do but it takes quite a bit of time and it is not something that we currently have our own need for and so we need to always find a way to fit this work between other tasks. If there are any parties that want to work with us on an optimized version of the codec we'd be happy to help them contribute to this, it would typically require knowledge of ARM assembly.

Andrew"

Anyone good at ARM assembly over here? lol

magdesign, 2019-11-21:
I think this guys would be able to write the code: 
https://github.com/z37soft/INetSyphon_RaspberryPi_SDK
http://www.techlife.sg/

langolierz, 2019-11-22:
Thanks for the update ! Def would be keen to intergrate this once it is
supported; a bit out of my depth to help with that but will keep an eye on
the progress :) , please let me know if you hear more !