cyberboy666/r_e_c_u_r #101: MIDI Enhancement: assignable macro CC mapping in recur

ParticlesIntoWaves, 2019-12-11:
Messing with my Kaoss Pad, and realizing how amazing it would be to control multiple shader parameters with one CC. A menu option to assign this in recur would be phenomenal. Reserve two MIDI CCs for "X" and "Y" (to keep with the Kaoss theme) and allow up to three shader parameters (lay0 x0 to lay2 x3, and what the hell add strobe while you're at it) to be assigned to each one in the menus.

NOW you can pull up on an Entrance :)

langolierz, 2019-12-11:
i dont quite get what you mean in this one - a kaoss pad is just two seperate cc channels for x and y no ? you want both channels to map to the same parameter or ?

ParticlesIntoWaves, 2019-12-11:
I was only referencing the Kaoss Pad as a use case example. Pitch bend and mod wheel work just the same for the sake of illustration.

The idea is to have macro control over multiple shader parameters simultaneously from a single MIDI CC. For example “Macro 1“ would receive only on CC13, but would alter 3 destinations in the shader modifiers, from lay0 X0 to lay2 X3. I think, because different combinations of shaders would require different combinations of parameters for practical macros, it makes the most sense for this to be available in the menus so it can be altered on the fly.

Having just one of those freely assignable shader macros would be pretty bonkers, having two would be stellar!

doctea, 2019-12-12:
The very basic use case here can already be done by editing the midi mapping file I think?  Possibly need some new actions adding too though.

This could be partly implemented with the 'custom plugin script' idea, if the custom scripts will be able to intercept MIDI messages and reroute them... 

However, what I think would also be necessary for this to work well (if I've understood it right) is to have a way to adjust the value mapping from the CC -> target parameter using a user-entered formula.  So for example, 

"Target A value = input" for straight mapping,
"Target B value = 1 - input" for inverse mapping,
"Target C value = 0.5 + input/2" to set the value range as 50-100%

(Similar to the way that you can map controller values with formulas in FL Studio)

I'm currently experimenting with some enhancements to the actions & MIDI mapping to support 'odd' controllers, if I'm successful I think I'll have a lot of these MIDI requests covered :)

ParticlesIntoWaves, 2019-12-12:
Yes, I figure one could effectively accomplish it by editing the map file to make one CC edit three parameters, but if the shader configuration changes then the macro may not be suitable anymore. Which is why I advocated for an option to map this inside the menus :)

The suggestion for how to interpret the input is great. A simple way to implement this would be a positive and negative offset to each parameter slot in the macro. That would allow bipolar manipulation with fine Or wide control for specific parts of it.

ParticlesIntoWaves, 2019-12-12:
You know... now I have to advocate for THREE macros because I just had another totally sick idea.

Again, use case scenario, but the KP3 can send three CCs from the touch pad, one each or X and Y, and a third for touch on/off. If I had three macros I could use the MIDI touch data to turn my shaders on or off when using the pad! Damn that would be so cool and would definitely blow away an entrancer ;) 

Damn it, you’re going to make me get into coding.

doctea, 2019-12-13:
The other requirement you're asking for here would be per-shader parameter mappings, I think?  or just a way to switch between different sets of mappings quickly would achieve it?

ParticlesIntoWaves, 2019-12-14:
I want to be able to program the individual assignments in the menu on the fly. I‘m imagining a macro menu tree in the settings that expands into three slots. Each slot can load any shader layer’s x#, and has modes for interpretation (positive, negative, offset, relative, catch up, etc).

doctea, 2019-12-14:
So sort of like a meta-preset mode?