cyberboy666/r_e_c_u_r #102: MIDI Enhancement: New default mappings

ParticlesIntoWaves, 2019-12-11:
Because I'm entitled... I think it would be a good idea to give any toggles and macros their own MIDI CC/note in the json map. I realize I could edit the file myself, but to me it's logical that these simply be included by default because it removes a very tedious and confusing step from the process for users and immediately opens up recur for deep mappings with external controllers.

Any function that requires a key macro, and many toggles in the menus could be added, such as:

X3 as speed on/off
Video bank select
Shader bank select
Feedback on/off
Shader lay0,1,2 on/off
etc. etc.

I'm sure this isn't high priority, however I do believe it would make recur more accessible from the get-go, and then no one ever has to mess with the json files to achieve this extra level of control.

langolierz, 2019-12-11:
im thinking , maybe a setting where you can toggle different mapping files ? then you or anyone else can make there own custom mappings (with all the flattening / macros etc they want) and everyone can share and choose whichever suits their controller / setup ?

ParticlesIntoWaves, 2019-12-11:
My thought was that the default map installed with recur should _also_ include a lot of the common FN macros or menu toggles individually mapped because it would effectively democratize the MIDI mapping process and make deep integration with MIDI devices relatively painless for anyone. If I need to help create this that’s great, but I’ll need lots of help :p

doctea, 2019-12-12:
There are already quite a lot of functions mapped to note_on MIDI messages in the default MIDI mapping, but I'm working on some enhancements in this area as not all features are currently exposed in a way that makes sense to MIDI map.  

I'm not familiar enough either with standard MIDI CC controls or r_e_c_u_r to know precisely what the best mapping of CC->function would be, but providing some default templates for different functions seems a good idea, just needs the ability to toggle mapping files and someone to write mapping files for different workflows/controllers!

@ParticlesIntoWaves : if you have any ideas on what the default mappings should be then please share :)

Using json as the file format for configs is also a little bit awkward as its not possible to add comments, maybe should think about switching to a different file format so that the config files aren't quite so daunting to edit by hand for those wanting to customise things?

ParticlesIntoWaves, 2019-12-12:
For any function that is a toggle, macro or otherwise activated with a MIDI note trigger, I think the MIDI note number assigned would be ultimately irrelevant, however I suppose the most logical arrangement is to be organized around middle C.

For any continuous controls, it makes sense to start at CC# 74, which on most synthesizers is the filter cutoff frequency. It may sound arbitrary but it’s one CC# that any MIDI user is sure to know. I would recommend avoiding the first 25 or so CC#s because things like program changes and after touch are sent here and that can confuse things. My r_e_c_u_r currently changes settings when I switch programs on my KP3 because r_e_c_u_r is accepting CC0 to edit shader layer 1.

ParticlesIntoWaves, 2019-12-12:
At any rate, a flattened MIDI map included in the default json would make this all painfully easy, for me at least.

langolierz, 2019-12-13:
do you have a suggestion for another file format ? i guess you can
comment in yaml ? but yah. i chose json coz i thought it would be more
human readable... but i get that its knida confusing, obvs a ui around this
would make it easier, actually have some ideas around building mappings
more easily ...

doctea, 2019-12-14:
I've just pushed code to my midi feedback branch that automatically loads a custom midi mapping file if one exists based on the device name, and activates the experimental feedback mode if an APC Key 25 is detected.  

Expanding this, if we collect midi mapping files from people with different controllers then this should make it easy to have 'plug n play' 'out of the box' support for different controllers and workflows.  although I guess it would still be good to be able to override that in the menus.

And if the file format is going to change then we should do that before getting people to send their configurations I guess...

I'm afraid I don't have much of a suggestion on file formats.  json would be pretty great tbh if only it supported comments!

what about using python itself for the config files?  would the basic syntax for defining variables and trees of lists be too confusing?  it would certainly make it easier to map ranges of notes/CCs to banks/slots without a lot of error-prone typing and duplication...

ParticlesIntoWaves, 2019-12-15:
Plug and play would be nice but it would be dependent on the controller’s ability to load matching profiles, since controllers can be programmed a number of different ways. If all useful r_e_c_u_r functions were individually MIDI mapped in the default json, I could whip together a custom profile on any of my controllers, tailored to r_e_c_u_r, in a couple minutes, which is how I’d use it any other context. A synth or MIDI device usually has uneditable CC receivers. A MIDI controller is mapped to send on those channels.

doctea, 2019-12-15:
> Plug and play would be nice but it would be dependent on the controller’s ability to load matching profiles, since controllers can be programmed a number of different ways. 

That's what the code in my experimental branch does, loads a different config file based on the detected name of the MIDI device.

>If all useful r_e_c_u_r functions were individually MIDI mapped in the default json, I could whip together a custom profile on any of my controllers, tailored to r_e_c_u_r, in a couple minutes, which is how I’d use it any other context. 

Which functions are currently missing from the default config that are preventing you from doing this?

Would a list of the commands available for mapping help?  There are likely some missing and maybe you can point out some of the blanks that need implementing.

>A synth or MIDI device usually has uneditable CC receivers. A MIDI controller is mapped to send on those channels.

MIDI controllers are often uneditable and non-standard too -- for example I've been using both an Axiom Air32 Mini and an Akai APC Key 25 and they each send out different ranges of CC values (CC#0 vs CC#48 for example), notes on different channels etc so it makes sense to have configs for different controllers.  I know the Akai APC Mk2 has its own quirky CC/note mappings too.

If you could write a mapping file that covers all the defaults you'd expect then I bet that could be integrated, however its difficult to know what functions you'd expect to find on which notes/CCs?

langolierz, 2020-03-12:
will create a new issue for greater midi controller / mapping handling in general , for now i have added some flattening to the default midi mapping as originally suggested

closed 2020-03-12 by langolierz