cyberboy666/r_e_c_u_r #97: Feature request: MIDI input smoothing/slew

ParticlesIntoWaves, 2019-12-11:
I would like to one day see recur able to smooth out MIDI input. Right now, shader edits made with a MIDI controller are a bit jumpy since it’s a stepped motion. Adjustable slew rate would make that all a lot more fluid.

On a side note, I think it would also be a good idea if the values were 1:1 with MIDI, so 0-127 instead of 0-99

langolierz, 2019-12-11:
i get what you mean, also though, its more 'stepped' than 127 steps because trying to save on resources : 
https://github.com/langolierz/r_e_c_u_r/blob/master/user_input/midi_input.py#L70

 i think if this step size was 1 or 0 it would look a lot smoother. you can experiment with changing this value on your machine if you like - also ofcourse pi4 might also help here ;)

ParticlesIntoWaves, 2019-12-11:
So what you’re saying is you can’t smooth it anymore because the input values of the shaders are low resolution? Or in other words, the shaders themselves do not have ultra fine control

langolierz, 2019-12-11:
No the shaders can take finer resolution, it’s just the the main loop is
not running fast enough to process more resolution without Lagging, or
something needs to be optimized first

langolierz, 2020-03-12:
i refactored the midi cc logic - it should be faster and smoother now than before - can reopen if still needed after testing new version, pending release

closed 2020-03-12 by langolierz

miroslawgwiazda, 2020-06-20:
I'm interested in the implementation of CV via midi in relation to clip/image changes, are you able to trigger changes synchronized to a beat? A simple example would be: one image triggered by a kick drum and another by a snare... Are load times fast enough for such an application? Any information would be much appreciated

langolierz, 2020-06-20:
👋 @miroslawgwiazda  

this is a good question.  just to clarify you want to trigger something on a midi input ? or a cv / gpio pin input ?

if using midi , then you could map your trigger event to a midi note on , and this can be mapped to trigger changing the sample etc. if you are trying to trigger from a modular directly, probably best wou;d be a small transistor circuit to a gpio pin on the pi. then would need to add some code to listen to change on this pin in recur and map to an action. i can help if you want to go down this route.. (another way is using the existing cv extension circuit and writing a small action function that changes the clip when a cv threshold is reached - although this depends on the length of the 'trigger' )

> Are load times fast enough for such an application?

this depends a bit, basically the 'switching' is fast , but only when the next player is already loaded, which takes a second or two. so triggering on every kick might be too fast , but on every bar probably will be ok. in the main demo video there is a part where i am triggering video changes using a midi sequencer per bar like this

miroslawgwiazda, 2020-06-20:
I apologize this is my first foray into Raspberry Pi so elements of the platform are somewhat new to me. I was unaware of the GPIO option, would I need something to interface this with my Eurorack, a patch bay for example? Additionally, could this interpret other forms of CV with the end result being able to navigate through a single video by way of a modulation source?