cyberboy666/r_e_c_u_r #98: Banks of 8 instead of 10

ParticlesIntoWaves, 2019-12-11:
Ok I may be the only person to really care about this, but many if not all MIDI controllers have their pads and buttons arranged in banks 8 or 16. With recur being banks of 10, it’s not very elegant to map my MIDI controller. With a 16-pad controller, it would be ideal to map a bank of 8 videos and 8 shaders to the pads for activation and switching.

I know this somewhat “breaks” the arrangement of the keypad mapping, but I’m all for reassigning those two extra keys to perform other functions! ;-)

langolierz, 2019-12-11:
it sounds like mostly a mapping problem to me no ? i mean there is nothing stopping you only using 8 slots and remapping the other keys to hotkeys or something else you might prefer (i guess technically the display would have a couple lines of real-estate you werent using , but still)

and yeah in general i will be thinking about how to make the mappings more accessible.

ParticlesIntoWaves, 2019-12-11:
I guess you’re right, I don’t have to use 10 slots LOL.

closed 2020-03-12 by langolierz