cyberboy666/r_e_c_u_r #99: UI Enhancement: Shortcut to “X3 as speed”

ParticlesIntoWaves, 2019-12-11:
Most shaders really benefit from the X3 as speed feature, but a lot of shaders also use X3 for important functions. Instead of going into the menu to turn this on or off constantly, a shortcut macro of some kind would be really helpful. Two ideas:

1) FN+number key toggles X3 as speed on or off
2) MIDI CC for X3 momentarily toggles speed on or off when FN is held down during editing (i.e., hold FN and turn MIDI knob to change speed)

ParticlesIntoWaves, 2019-12-11:
Another idea is a MIDI note can toggle X3 as speed on and off, as I suggested in the new MIDI mappings request.

langolierz, 2019-12-11:
the way the system is designed it is actually very easy to add hotkeys for setting actions like this. this would only be a few lines to make happen...

i might have to help you get your hands dirty lol

ParticlesIntoWaves, 2019-12-11:
I guess this kind of lumps into having more of the functions individually mapped in the default json file so they can be quickly accessed with an external controller.

doctea, 2019-12-12:
Suggestion 2 "MIDI CC for X3 momentarily toggles speed on or off when FN is held down during editing (i.e., hold FN and turn MIDI knob to change speed)" is an interesting idea.  Maybe could be implemented in a generic way by transposing all having a separate mapping on a higher CC for the alternate feature and having all CC numbers transposed +64 if FN is held.  Or transposing MIDI Channel number when FN is held or something like that

doctea, 2019-12-14:
@ParticlesIntoWaves I think I've got your requests covered in my two pull requests above.  Let me know what you think and if you want any help trying them out :)

ParticlesIntoWaves, 2019-12-14:
How do I go about using that?

doctea, 2019-12-14:
> How do I go about using that?

what kind of access do you already have to your recur?  do you know how to get to a console on it to update some files?  are you familiar with git at all?  i'll try and help talk you through it