cyberboy666/r_e_c_u_r #112: Building outside of Pi / Development Docs

averagehat, 2020-01-21:
It would be great to be able to build this project, and/or https://github.com/langolierz/c_o_n_j_u_r/ and/or https://github.com/langolierz/ofxVideoArtTools/ outside of the ARM architecture. 

The big problem, at least with conjur/the ofx addon, seems to be the OMX dependency.

Being able to build in an emulator, or VM, or docker, or something would be helpful for development times. Has anybody have any ideas or experience trying?

Also, I can use some information for 'development' docs; so let me know what you've found useful for developing on the Pi 3 itself; particularly compile times, maybe you've installed libclang for editor support, I don't know.

langolierz, 2020-01-22:
This is a good one , actually a main motivator of abstracting out to
ofxVideoArtTools was to make the backend more cross platform ,
ofxVideoArtTools can switch between pi specific packages (for optimized
performance etc ) and ones that should work on a Mac etc.

I’m not sure how well it’s working like that right now but that was the
idea atleast , maybe I can try get this working / show an example or two ..

averagehat, 2020-01-22:
Cool. Right now c_o_n_j_u_r also has a direct dependency on omxplayer, I
think
I was not able to get omxplayer to build anywhere outside of a Pi

langolierz, 2020-01-22:
It shouldn’t , you can tell a recurVideoPlayrr what type to be , here :
https://github.com/langolierz/c_o_n_j_u_r/blob/master/src/ofApp.cpp#L25

, I think when the backend is set to ofPlayer it uses the default
openframeworks video player ,
Although maybe you then need to remove ofxOmxplayer from the addon.make also