cyberboy666/r_e_c_u_r #133: Trouble with LPD8

MiddleJanuary, 2020-04-06:
Hello!

Big fan of the project! I have my r_e_c_u_r built, it does a lot of cool things!

I have an LPD8 connected to it and mapped out some midi functions.

CC #0 does not seem to work well with the LPD8.

I mapped the top 4 control knobs on the LPD8 as cc#0, cc#1, cc#2 and cc#3.

CC #1-3 work perfectly fine, it's cc#0 where I am having trouble.

When I turn the knob nothing happens. Does anyone else have this problem?

It works in the demo videos, not sure if I'm missing a line of code somewhere..

Or maybe I have a busted LPD8. Still running version 2.0.2, any info would help.

Cheers!

langolierz, 2020-04-06:
hii ! glad you are enjoying the project !

ahh.okay - the lpd8 is the same controller that i use to have (i broke the
usb-port off, still it sits in a pile waiting to be fixed..) , but yes i
guess it was working for me back then. 2 things i can think of that might
narrow the problem:

1 - what if you try mapping the top 4 control knobs to cc#3, cc#2, cc#1,
cc#0 ? if it was a problem with your controller i would expect now that
cc#3 would not work while the others do (same knob) , if not then its most
likely a problem with recur.

2 - maybe there is an off-by-one error where the code is reading cc#1 as
input 0 and cc#2 as input 1 and cc#3 as input 3 ? and this is why the first
knob does nothing ? to test this you could try mapping the top 4 controls
to cc#1, cc#2 and cc#3, cc#4 ?

if after trying both these things and problem is the same, could be a bug
in the code, we can try figure out.
lemme know how it goes !

MiddleJanuary, 2020-04-07:
Wow! Awesome trouble shooting!

Map 1: cc#3, cc#2, cc#1, cc#0
The first three knobs turn parameters 3, 2 and 1.
Nothing happens when I turn the 4th knob.

Map 2: cc#1, cc#2 and cc#3, cc#4 
The first three knobs turn parameters 1,2 and 3. 
The 4th knob does nothing, parameter 0 doesn't move at all.

This is fun! Very cool project!

Cheers!

MiddleJanuary, 2020-04-07:
Oh side note! Assigning cc#0 to a pad, then pressing the cc function on the LPD8 and hitting the pad, r_e_c_u_r does recognize the midi information and moves parameter 0 according to the velocity the pad was struck. Softer for lower numbers, harder for higher.

If it is a bug in the system, for some reason it only happens with the potentiometer value and not the pad.

Hope this helps!

Cheers!

langolierz, 2020-04-07:
That is very helpful , thanks. Well the good news your controller doesn’t
seem to be physically broken. But yeap it seems that the cc0 command is not
being interpreted correctly.

Others have told me there is problems with using cc0 because this usually
means LSB for a program change -
http://www.tweakheadz.com/midi_controllers.htm

Still I would like to know what exactly is being sent when you turn a knob
mapped to cc0 and how it is different to a pad mapped to cc0 !

I will take a look in the lpd8 mapping software and let you know what I find

MiddleJanuary, 2020-04-08:
I keep on thinking about a simple midi controller interface..

A lot of midi controllers have 8 or 16 pads, I've been thinking about a setup that would remove the need to include the 10 key pad.

Woke up early this morning and had to type out this 4 x4 setup..

   Normal setup                                                         Function Key enabled
   _________    _________    _________    _________	   _________    _________    _________    _________  
   |             |   |             |   | select |   |              |         |            |    |             |   |slctshdr|   |             |
   |     <      |   |     >     |   |     ▇    |   |  dsply   |	   |      [     |   |      ]      |   |     ▇     |   |    —>   |
   |________|   |________|   |________|   |_________|	   |________|   |________|   |________|    |________|
   _________     _________    _________    _________	   _________    _________    _________    _________
   |             |   |      ◯     |   |             |   |             |	   |             |   |             |   |            |    |             |
   |     fn     |   |    rec    |   |      0     |   |      1     |	   |     fn     |   |    rec   |   |   clr [   |   |    clr ]   |
   |________|   |_________|   |________|   |________|	   |________|   |________|   |________|   |________|
   _________    _________    _________    _________	   _________    _________    _________    _________
   |             |   |             |   |             |   |             |	   |             |   |             |   |             |   |             |
   |      2     |   |     3      |   |     4     |   |      5     |	   |prv bnk|   |nxt bnk |   | clr bnk |  | mirror  |
   |________|   |________|   |________|   |________|	   |________|   |________|   |________|    |________|
   _________    _________    _________    _________	   _________     _________    _________    _________
   |             |   |             |   |             |   |             |	   |             |    |             |   |             |   |              |
   |     6      |   |      7     |   |     8     |   |      9     |	   | shader |   | frames |  |feedbck|  |shutdwn|
   |________|   |________|   |________|   |________|	   |________|    |________|   |________|   |_________|

I also have an idea for a midi map where the pad are in a 2 x 8 setup, like the Arturia Beatstep.

     Normal Setup
     _________   _________   _________   ________    ________    ________    ________   ________  
     |            |   |            |   | select |   |           |   |            |   |           |   |           |   |           |
     |     <     |   |     >    |   |     ▇    |   |     0    |   |     1     |   |    2     |   |    3     |   |    4     |
     |________|   |_______|   |________|   |_______|   |_______|   |_______|   |_______|   |_______|
     _________    ________    ________    ________    ________    ________   ________    ________
     |             |   |    ◯    |   |           |    |           |   |           |   |           |   |           |   |            |
     |     fn     |   |   rec   |   | dsply |   |     5     |  |    6     |   |    7     |   |    8     |   |     9    |
     |________|   |_______|   |_______|   |________|  |_______|   |_______|   |_______|   |_______| 

     Function Key Enabled
     ________    ________    ________    _______	_________    _________    _________    ________
     |           |   |           |    |           |   |           |	|             |   |            |   |              |   |            |
     |     [     |   |     ]     |   |    +     |   |  —>   |	|prv bnk|   |nxt bnk|   | clr bnk |   | mirror |
     |_______|   |_______|   |_______|   |_______|	|________|   |________|   |________|   |________|
     ________    ________    _______     ________	 ________     _________    ________     _________
     |           |   |           |    |          |   |            |	|             |   |             |   |             |   |             |
     |    fn    |   |   rec   |   |  clr [  |   |   clr ]  |    |shader |   |frames |   |feedbck|  |shutdwn|
     |_______|   |_______|   |_______|   |_______|	|________|   |________|   |________|   |________|

With 16 encoders the Arturia Beatstep can easily include the 3 sets of shader parameters.
Leaving 4 encoders left over for additional control, maybe volume control? Or something else!

My initial plan was to get a second LPD8, so I could have 16 pads and 16 knobs. 
But then I realized the Function Key wouldn't carry over to the second LPD8.
However, on the LPD8 is a selection to switch pads between cc, note_on and program change.
My new idea is to edit the code to add cc numbers to the functions with note_on values.
Now that I'm thinking about it.. I would need a flattened MIDI control map with all FN+ combos given a dedicated cc#.

That seems like a lot of work.. I guess I'll just buy an Arturia Beatstep instead.

Let me know what you think!

MiddleJanuary, 2020-04-08:
I keep on thinking about a simple midi controller interface..

A lot of midi controllers have 8 or 16 pads, I've been thinking about a setup that would remove the need to include the 10 key pad.

Woke up early this morning and had to type out this 4 x 4 setup..

  
     Normal setup
     _________  _________  _________   ________        
     |       |  |       |  | select|  |       |       
     |   <   |  |   >   |  |   ▇   |  | dsply |       
     |_______|  |_______|  |_______|  |_______|       
     _________  _________  _________   ________          
     |       |  |   ◯   |  |       |  |       |       
     |  fn   |  |  rec  |  |   0   |  |   1   |       
     |_______|  |_______|  |_______|  |_______|          
     ________   ________   ________   ________     
     |       |  |       |  |       |  |       |
     |   2   |  |   3   |  |   4   |  |   5   |
     |_______|  |_______|  |_______|  |_______|
     ________   ________   ________   ________
     |       |  |       |  |       |  |       |
     |   6   |  |   7   |  |   8   |  |   9   |
     |_______|  |_______|  |_______|  |_______| 

     Function Key enabled
     _________  _________  _________   ________        
     |       |  |       |  |slcshdr|  |       |       
     |   [   |  |   ]   |  |   ▇   |  |  —>   |       
     |_______|  |_______|  |_______|  |_______|       
     _________  _________  _________   ________          
     |       |  |       |  |       |  |       |       
     |  fn   |  |  rec  |  | clr [ |  | clr ] |       
     |_______|  |_______|  |_______|  |_______|          
     ________   ________   ________   ________     
     |       |  |       |  |       |  |       |
     |prv bnk|  |nxt bnk|  |clr bnk|  |mirror |
     |_______|  |_______|  |_______|  |_______|
     ________   ________   ________   ________
     |       |  |       |  |       |  |       |
     |shader |  |frames |  |feedbck|  |shutdwn|
     |_______|  |_______|  |_______|  |_______|

I also have an idea for a midi map where the pad are in a 2 x 8 setup, like the Arturia Beatstep.

     Normal Setup
     _________  _________  _________   ________   ________   ________   ________   ________  
     |       |  |       |  | select|  |       |  |       |  |       |  |       |  |       |
     |   <   |  |   >   |  |   ▇   |  |   0   |  |   1   |  |   2   |  |   3   |  |   4   |
     |_______|  |_______|  |_______|  |_______|  |_______|  |_______|  |_______|  |_______|
     _________  _________  _________   ________   ________   ________   ________   ________  
     |       |  |   ◯   |  |       |  |       |  |       |  |       |  |       |  |       |
     |  fn   |  |  rec  |  | dsply |  |   5   |  |   6   |  |   7   |  |   8   |  |   9   |
     |_______|  |_______|  |_______|  |_______|  |_______|  |_______|  |_______|  |_______|

     Function Key Enabled
     _________  _________  _________   ________   ________   ________   ________   ________  
     |       |  |       |  |slcshdr|  |       |  |       |  |       |  |       |  |       |
     |   [   |  |   ]   |  |   ▇   |  |  —>   |  |prv bnk|  |nxt bnk|  |clr bnk|  |mirror |
     |_______|  |_______|  |_______|  |_______|  |_______|  |_______|  |_______|  |_______|
     _________  _________  _________   ________   ________   ________   ________   ________  
     |       |  |       |  |       |  |       |  |       |  |       |  |       |  |       |
     |  fn   |  |  rec  |  | clr [ |  | clr ] |  |shader |  |frames |  |feedbck|  |shutdwn|
     |_______|  |_______|  |_______|  |_______|  |_______|  |_______|  |_______|  |_______|

With 16 encoders the Arturia Beatstep can easily include the 3 sets of shader parameters.
Leaving 4 encoders left over for additional control, maybe volume control? Or something else!

My initial plan was to get a second LPD8, so I could have 16 pads and 16 knobs. 
But then I realized the Function Key wouldn't carry over to the second LPD8.
However, on the LPD8 is a selection to switch pads between cc, note_on and program change.
My new idea is to edit the code to add cc numbers to the functions with note_on values.
Now that I'm thinking about it.. I would need a flattened MIDI control map with all FN+ combos given a dedicated cc#.

That seems like a lot of work.. I guess I'll just buy an Arturia Beatstep instead.

Let me know what you think!

>

langolierz, 2020-04-08:
yes. this is something i think about also. currently i am using a korg
nanokontroller , which has enough buttons to not need the keypad , and
enough knobs to control all the params at once. (and it cost the same
roundabout as lpd8 , thats why i bought this when my lpd8 broke)

you can see how i have it setup here :
https://github.com/langolierz/r_e_c_u_r/wiki/default_midi_mappling

also in the newest release you can use cc values for functions note_on.
although recur doesnt support 2 midi controllers at once yet (although have
been thinking about this recently) and i really want to make it easier for
people to edit the midi map so they can build ideas like this tho -

to solve your original problem , if you can edit the midi map on recur then
the easiest thing would be to just change the params cc values to #1, #2,
#3, #4 . i. am quite stumped why the cc0 doesnt work. would have to get you
to run some commands on the pi to see what is really happening there.

MiddleJanuary, 2020-04-08:
That's a cool setup!

Good to know that the newest release has cc values for note_on functions.

Didn't realize that recur does not support 2 midi controllers at once, that makes sense now.
Ruins my plans on buying a second LPD8 tho.. Damn that would have been a compact setup!
Came up with a list of midi controllers below or around $100, if any one is interested.

Akia LPD8

Akia MPD218 

Akia MPK Mini MkII

Arturia Beatstep

Behringer CMD DC-1 (out of stock, still on eBay)

Behringer X-Touch Mini Ultra-compact Universal USB Controller‎

Korg padKONTROL

Korg nanoKONTROL2

Korg nanoPad 2

Native Instruments Maschine Mikro MKII MIDI Controller

Nektar Impact LX25+ 25-key Keyboard Controller

Samson Conspiracy MIDI controller

I'm sure there are others but those are just some personal favorites.
Also found a really cool DIY build on Adafruit, the Mini UNTZtrument: 3D Printed MIDI Controller.
Think I might build one of these, but add 8 more potentiometers.

Going back to the original problem, editing the midi map seems like the only solution.
Since I have to update to version 2.1.0, this would be a good time for that.

I am also still curious to why cc#0 doesn't work and am absolutely willing to run some commands on the pi if you can figure out what is happening. I was reading through the Facebook group and it looks like a handful of people are having trouble with cc#0, so if I can be of any help let me know.

Cheers!

MiddleJanuary, 2020-04-09:
I can't stop thinking about this..

16 pads, 16 knobs, an xy pad..   and audio input..

I'm thinking about making a pcb for an easy, inexpensive diy midi controller build.
I will definitely share the gerber files here once I'm done.

Looking for a simple xy pad solution, I realized v2.1.0 uses the touch screen.
Could be a possibility to assign a button to switch between modes, the settings menu and xy pad..

I also saw this software for $5 dollars!.. TouchOSC by Hexler Limited.
Could use a smart phone as an xy pad, or even create a whole 16pads/16knobs/xypad setup..

I know guergana is working on the r_e_m_o_t_e project. Maybe this could work there..
And I know doctea is working on "Sound reactivity [experimental]" I'll have to dig into those notes.

Again, I am also still curious to why cc#0 doesn't work and am absolutely willing to run some commands on the pi if you can figure out what is happening. If I can be of any help let me know!

I LOVE THIS PROJECT!!

langolierz, 2020-04-10:
im really glad you are excited about it - that was one of my main goals
when starting , to empower people to be part of creating the art-making
tools they want to use.

i also have been day-dreaming about making a little custom controller for
recur (although much smaller than what it sound like you have in mind) ,
the main thing stopping me is that the korg-nano does do almost every ting
i need now anyway , so hard to justify too much time into this but... yes
please share anything you come up with

re the xy-pad ; i looked into this i alil. it seems like you can get some
replacement touch pads for laptops to interface with arduino (
https://www.instructables.com/id/Arduino-Laptop-Touchpad/) , which could be
a place to start. and you are right , a cellphone is also a good source of
xy-action. the latest version supports osc so you can use touchOsc with it
already (ibought the software few months ago - it is nice), and there can
def make an xy-pad. and also plan to add this to the remote extension.

for sound reactivity, there is someone in the group who has made a custom
midi control with soundreactivity. they used something like  this i think
https://www.sparkfun.com/products/13116 , or another option is to use a
usb-audio device  like what Tristan (doctea) was working on. i would like
to work on this feature soon too because i already have some usb-capture
cards that input video to recur , would be nice to also get sound on them
aswell.

finally - thanks a lot for being willing to hunt this cc0 problem. i will
write a small python file on my pi that prints out the incoming commands
and then send it to you to run on yours and we can try spot the difference

hamsterwoede, 2021-06-25:
I have been setting up my LPD8 with r_e_c_u_r today, and found out that for some reason it just doesn't send anything on cc0. At least as far as I can see, nothing arrives on OSX.  I have had it working in the past with a previous version. But I might have also updated the firmware in the meantime. As far as I know it is a problem with the LPD8, and not so much r_e_c_u_r. Annoying really.