cyberboy666/r_e_c_u_r #143: MIDI CC to rate?

TubularCorpration, 2020-08-06:
I've been poking around in the video player code a bit and it looks like it wouldn't be that tough to implement CC control of playback rate, but I'm not sure what the valid range of speed values are. I think I could probably work out how to scale CC data to a valid range and write an action that would read the current playback location, set the rate to the scaled CC value, and then retrigger the clip from the current playback location that was read earlier (from what I can tell the current increase/decrease speed functions have to retrigger the current clip to make the speed change take effect, so I'm assuming if I wanted to change the speed on the fly without changing the current playback location the function would need to retrigger it at the current location after setting the new rate).  Does this sound correct?  

I'm basically looking for a way to set the speed to a specific value via MIDI in the same way I've been setting the playback position, so the sequencer I'm using to control playback position can use a different CC to control speed. Not expecting smooth changes, just looking to fire off single CC values.

Also when I was in there I noticed a typo in actions.py:

    def decrease_speed(self):
        print("increasing speed !")

langolierz, 2020-08-07:
Re the video playback speed...

I thought it would be possible to implement this too. Actually I spent a
very long time trying to get this working (hence all the code /comments
Around the place for this haha) but it never really worked that well ...
only sometimes for some videos etc. you can try uncomment those parts and
play around but yah it can be a bit frustrating. Also this is partly why I
started working on detour the frame sampler... but yeah I know it’s not
quite the same

TubularCorpration, 2020-08-07:
Probably not worth the effort then.  Speaking of intermittent, using low resolution CC LFOs was working great two days ago, and then the next day with the same settings and clip it didn't work reliably at all, even the slowest setting was more than it could handle, so who knows.  I'm getting more than enough variation just with CC to position on mutiple tracks, once I have a few long chains built I can use one track to change clips and a couple of other tracks to change playback position and end up with a lot of semi-generative variety, and then reserve LFOs for controlling shaders when I start using those.

Detour actually seems really promising, I'm just trying to focus on one area at a time as I figure out how I want to fit it in to my workflow.