cyberboy666/r_e_c_u_r #145: Hide files and folders beginning with a dot from file browser

joem, 2020-08-16:
I think it'd be great if files and folders beginning with a dot were hidden from the file browse, since in Linux and macOS those are considered hidden files, and I think Windows doesn't really like those types of files much (maybe? can't remember).

This would be especially helpful for macOS users, since the OS can leave various dot-files around on USB keys, including ones with the filename of the video files in it, so they end up showing up in r_e_c_u_r's file browser. (example: if you have a file "movie.mp4" macOS can also create a file called "._movie.mp4")

doctea, 2020-08-16:
I thought this was already in the latest versions (may need to 'update code' if you haven't updated before). Let me know if there's something I missed though..

joem, 2020-08-19:
Ah yeah, after updating the code that seems to be working. Ha. Sorry I missed that!

closed 2020-08-19 by joem