cyberboy666/r_e_c_u_r #146: new control scheme (for default?) / nanokontrol

langolierz, 2020-09-08:
it has been suggested to me from @ParticlesIntoWaves that this would be an ideal control scheme for using recur with nanokontroller (and not needing to rely on the keypad which can be a bit hacky)

[image]

> An alternative idea for the 1–8 buttons
-Top row is video slots
-Second row is video banks 
-FN+second row buttons is shader banks
I realize r_e_c_u_r can do more videos and banks than this but 8x8 is the most elegant for NK2... and is still plenty of videos

on the whole i dont think it would take much to implement this. 

i have a few open questions like whether we should create 'breaking changes' with the default layout or add a setting option to choose from different layout schemes...

besides that i think everything thing else will be possible by editing the mapping json

Licoricious, 2020-10-13:
I think a setting option to choose between this and the current control scheme would be excellent. I definitely think this layout would work better for controlling _recur_, but I also enjoy having some assignments free for me to control other gear simultaneously. The default scheme should probably remain the same, at least for now ("WE DO NOT BREAK USERSPACE"-LT). Down the road, I think it wouldn't be too hard to implement a MIDI learn mode and the ability to save various control schemes--but that's getting ahead of ourselves at the moment.

doctea, 2021-03-12:
I might be biased ;-) but I'd like to see this done using the controller-specific mapping stuff I implemented for the APCKEY25.  That way you could have this be the default control layout for the Nanocontrol, but keep the 'old' mapping for other controllers, and expand in future with layouts that make sense for other controllers too.