cyberboy666/r_e_c_u_r #147: LOOP_TYPE parallel sampler issues

Licoricious, 2020-10-13:
I was playing around in parallel mode tonight, and noticed some irregularities in the loading behavior.

1. When playing a video in the NOW player, with the NEXT player selected (cyan bar at top) and next.player.toggle_show() set to false:

When you load a video into the NEXT player,  next.player.toggle_show() is set to TRUE. This causes switch_to_next_player to toggle_show() for both the NOW and NEXT players to FALSE--hence dropping video feed.

2. When playing a video in the NEXT player, with the NOW player selected (yellow bar at top) and now.player.toggle_show() set to false:

When you load a video into the NOW player, now.player.toggle_show() is set to TRUE. This causes playback to jump to the NOW player, even if that was unintended.

It's pretty clear these things are two manifestations of the same bug.
I have a guess as to the cause of these issues (line 136 in actions.py), but I wanted to make a post, as I have no experience in 
Python or with Github in the past, so it may take a bit for me to fix this on my own.

My apologies if this behavior is working as intended or I'm misunderstanding something.

langolierz, 2020-10-13:
thanks for the heads up @Licoricious !

that does sound strange, and the code on line 136 looks a bit off too. most likely it is a bug (or some weird thing i was trying and forgot to fix)

i would have to try it myself to check, but meanwhile if you want to try edit the code yourself to do something that makes more sense i can help you get set up with that. maybe can then merge your changes into main if it is better.

Licoricious, 2020-10-14:
So commenting out that line fixes situation 1 above, albeit in a somewhat hacky way. It ends up switching channels simply by toggle_show()-ing the NOW player. This works since the NOW player appears above the NEXT player. 

Situation 2 is a bit trickier. The root of the problem seems to be that reloading the player sets toggle_show() to TRUE, when you really want to leave it in whatever state it was before. I tried added another line to turn it back off, but due to loading time delay the behavior is highly inconsistent. I'm gonna keep messing around and see if I can find a simple solution to this.

langolierz, 2020-10-14:
thanks so much for looking into it (and happy you are figuring it out !)
when im on recur next i will also take a look and let you know what i find