cyberboy666/r_e_c_u_r #149: White background on LCD

bsom, 2022-07-22:
Any idea why on my freshly built r_e_c_u_r using the 2.1.2 image the background of the LCD is white instead of black?

octopus-arts, 2023-07-08:
Same here + the displayed output is upside down relative to the HDMI and composite outputs.

octovolt, 2023-10-28:
This might be a Waveshare firmware issue. My version of the LCD is a Waveshare 3.5inch RPi (B), which is giving me inverted colors. I'm still trying to figure out how to fix it, but I'm hopeful that this discussion will point me in the right direction: https://raspberrypi.stackexchange.com/questions/113522/3-5-inch-rpi-lcd-with-colors-problem

I was able to rotate the display in config.txt by changing this line:
```
dtoverlay=waveshare35a:rotate=270
```
to this:
```
dtoverlay=waveshare35a:rotate=90
```

cyberboy666, 2023-10-28:
thanks for this info ! 

i have heard a few reports from people about this inverted display

easiest thing to do i think would be to add an option in the settings that lets you invert the display in the python code.. shouldnt be hard to do and also as bonus maybe some people prefer the white background version in some context...

im not really releasing new versions of this project any more - but can try find time to figure out how to do this inverting since its a request i get every so often.

also while here could add a `flip display` settings option too - which would do it the same way you describe here with the dtoverlay

mrfugu, 2023-10-28:
Still ppl out here that would love to see rpi4, or 5 builds, thinking/assuming that the video and camera delay times could be greatly reduced…. But completely understand why one would want to avoid  going down the rabbit hole again with an old abandoned project, Cheers!!Mrfugu Sent from my iPhone

jfre1, 2025-01-18:
The fix is using the different overlay and calibration. 

When you use the script as statet in the waveshare display manual, it overwrites the config.txt, but also overwrites other things like cmdline.txt and xorg, fbcp:
https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)

Here is the script code:
https://github.com/waveshareteam/LCD-show/blob/master/LCD35B-show-V2

I used this and changed things back after that, but it's maybe better, to just copy the overlays for 35-b-V2 from here:
https://github.com/waveshareteam/LCD-show/tree/master/boot

and put them in the boot folder on the pi.

Then you have two options: 

### 1) a short and simple option (quick and dirty), but you can't use the dev mode anymore:

change in config config.txt

from
`## switch for enabling lcd screen (the next line is being used even if its commented out)`
`##no_waveshare_overlay`

or (depends if you are in dev mode or not)
`## switch for enabling lcd screen (the next line is being used even if its commented out)`
`dtoverlay=waveshare35a:rotate=270`

to 
`## switch for enabling lcd screen (the next line is being used even if its commented out)`
`dtoverlay=waveshare35b-v2:speed=38000000,rotate=90`

The speed is to change the SPI speed, so you get more frames. Every display is different, so you can experiment, if you can go higher. For me, 38000000 was maxed out before the colors are glitching around.

### **2) the "advanced" option, to have all functionallity, but it's more time consuming:**

_(I use my display upside down to have the i_n_c_u_r controls above the screen, but if you want to have it below, you have to change the code to `rotate=90` instead of `rotate=270` )_

At first, you have to do the quick version too, but after that, there are some more things to do: 

There are two scripts in the dotfiles, you have to change:
https://github.com/cyberboy666/r_e_c_u_r/blob/master/dotfiles/switch_display_to_hdmi.sh
from
`sudo sed -i "s/##no_waveshare_overlay/dtoverlay=waveshare35a:rotate=270/g" /boot/config.txt`
to
`sudo sed -i "s/dtoverlay=waveshare35b-v2:speed=38000000,rotate=270/##no_waveshare_overlay/g" /boot/config.txt`

Next, there is also the counterpart:
https://github.com/cyberboy666/r_e_c_u_r/blob/master/dotfiles/switch_display_to_lcd.sh

`sudo sed -i "s/##no_waveshare_overlay/dtoverlay=waveshare35a:rotate=270/g" /boot/config.txt`
to
sudo sed -i "s/##no_waveshare_overlay/dtoverlay=waveshare35b-v2:speed=38000000,rotate=270/g" /boot/config.txt

but we are not finished yet! In the actions.py:
https://github.com/cyberboy666/r_e_c_u_r/blob/master/actions.py

Find the "switch_display_to_hdmi" function. For me it's line 729, but i've editet some things and I'm on the 2.10 image, so it could be a other line for you. It should look like this:

`def switch_display_to_hdmi(self):
        with open('/boot/config.txt', 'r') as config:
            with open('/usr/share/X11/xorg.conf.d/99-fbturbo.conf') as framebuffer_conf:
                if 'dtoverlay=waveshare35a,rotate=270' in config.read() and 'dev/fb1' in framebuffer_conf.read():
                    self.run_script('switch_display_to_hdmi')
                else:
                    self.message_handler.set_message('INFO', 'failed to switch display')`

then change it to:

`def switch_display_to_hdmi(self):
        with open('/boot/config.txt', 'r') as config:
            with open('/usr/share/X11/xorg.conf.d/99-fbturbo.conf') as framebuffer_conf:
                if 'dtoverlay=waveshare35b-v2:speed=38000000,rotate=270' in config.read() and 'dev/fb1' in framebuffer_conf.read():
                    self.run_script('switch_display_to_hdmi')
                else:
                    self.message_handler.set_message('INFO', 'failed to switch display')`

I'm not sure if i forgot something, because i changed it like two months ago. I don't use the touchscreen, but for me it's like inverted. Maybe there is something to change in the calibration, but i don't use it anyway.

There is also an interesting project, that's getting more fps out of the waveshare screens (mostly for gaming), but i don't think it's know if it's possible to use this for r_e_c_u_r and is maybe hard to implement:
https://github.com/juj/fbcp-ili9341
https://www.youtube.com/watch?v=dqOLIHOjLq4