cyberboy666/r_e_c_u_r #151: Unlimited sample / shader slots

tandy-1000, 2022-10-23:
Is there a hard limit on the number of sample / shader slots?
I've made a custom mapping for the Launchpad Mini, adding mappings for 64 slots in total. I haven't been able to test this mapping properly yet.

Is there functionality in the code to support this?

cyberboy666, 2022-10-23:
thats a good question.

in the case of samples you can already have unlimited slots. they are broken out over 'banks' (each of 10 slots) that are displayed on the screen. but you can have as many as you like

if you had (for example) 7 banks full of 10 samples each it would be possible to write a mapping that triggered each slot with a press of your launch pad button

you would just need to write a small helper function in `actions.py` that takes the bank and slot number from the mapping - (something like this might do it although i havnt tested):

```python
def _load_this_slot_and_bank_into_next_player(self, slot, bank):
    if len(self.data.bank_data) <= bank:
        self.message_handler.set_message('INFO', "bank %s doesnt exist" % bank)
    else:
        self.data.bank_number = bank
        self._load_this_slot_into_next_player(slot)
```
and then all you need to do is write the specific functions that are set in the json mappings:
```python
def load_slot_0_bank_3_into_next_player(self):
    self._load_this_slot_and_bank_into_next_player(0,3)
...
```

the case for unlimited shaders is a little bit more tricky since they dont have _unlimited_ banks like samples do. shaders have 3 _layers_ (which interact with each other and are a bit more 'hardcoded' into the backend) - it would be possible to increase number of layers although requires more changes. the other option would be to increase the size of each layer - the only reason it is 10 is coz thats what can fit on the display... maybe a way to 'scroll down' the layer on the display would make sense in this case. or a quicker more hacky approach would be just to hard-code all the shaders into one big layer that you cant see or change. but can switch between with ur launchpad...

tandy-1000, 2022-10-24:
I see, thanks for the quick reply, seems like I'll have to explore this a bit more.
If python had meta-programming features like nim that would make the specific functions a lot easier ...
I understand now, the way I understood it was that shaders were "played" similarly to samples, that makes sense.
Thanks!