cyberboy666/r_e_c_u_r #152: emulator

konsumer, 2024-09-09:
Hi, I made a simple lil script to send "knobs" to glslViewer, for shader-dev. It's very handy, since the uniforms are all the same, and you can use webcam or video-files with it, and it updates on save.

[image]

Feel free to use it directly here, or link to it, if you want:

https://github.com/konsumer/glsl_sandbox

cyberboy666, 2024-09-10:
so cool - thank you very much for this ! i will add links in shader dev wiki page

closed 2024-09-10 by cyberboy666

konsumer, 2024-09-10:
I am working on a [web-based version](https://github.com/konsumer/recurweb) to simplify installation (no need for python, glsviewer, etc on host, just go to a web-page.) I will update here when it's working.

konsumer, 2024-09-12:
Ok, I have [this](https://konsumer.js.org/recurweb/) setup. I am still trying to troubleshoot video-textures, but it works great with still images.